hubmartin/WS2812B_STM32F4 #3: Can't get working on F4 nucleo with RTOS

alxhoff, 2017-08-22:
Hi,

firstly, great library. I'm trying to get it going on an STM32F429ZI that is running FreeRTOS. I have followed your readme and the demo code but I cannot seem to get it going, i'm also quite new to embedded systems so not 100% sure how to check if all the DMA stuff is working etc.

I haven't touched my original config in CUBE which had no timers enabled or DMA stuff added. Looking at the init code it seemed to all get handled by the library.  I copied the visInit and visHandle functions from the visEffect example files. I have a single set of 3 LEDS on my gpio PE7 so in the ws2812.h I modified the defines to be as follows:

`// GPIO enable command
#define WS2812B_GPIO_CLK_ENABLE() __HAL_RCC_GPIOC_CLK_ENABLE()
// LED output port
#define WS2812B_PORT GPIOE
// LED output pins
#define WS2812B_PINS (GPIO_PIN_7) //(GPIO_PIN_0 | GPIO_PIN_1 | GPIO_PIN_2 | GPIO_PIN_3)
// How many LEDs are in the series - only valid multiples by two
#define WS2812B_NUMBER_OF_LEDS 3
#define WS2812_BUFFER_COUNT 1
#define SETPIX_4	// Fastest copying using bit-banding
`
I changed visHandle to:

`void visHandle()
{

	if(ws2812b.transferComplete)
	{
		// Update your framebuffer here or swap buffers
//		visHandle2();
		visDots(frameBuffer2, sizeof(frameBuffer2), 50, 40);

		// Signal that buffer is changed and transfer new data
		ws2812b.startTransfer = 1;
		ws2812b_handle();
	}
}`

I have a task that is running every 100 miliseconds and running visHandle().

It compiles and runs so I am not sure what I am missing as the setup described in your readme and the example seems very simple.

Any feedback would be great!

Thanks again,

Alex

hubmartin, 2017-08-22:
Hello Alex,

it should work :) I too don't think that RTOS could interfere with my
driver. What I recall is that I had some issues when the LED strip was too
short, it also has to have set even LED count. Try to set
WS2812B_NUMBER_OF_LEDS to let say 8 or the original value which is in the
example.
Also you have to enable clock for GPIOE. In your code you have forget to
change GPIOC to GPIOE, unless you enable that clock somewhere else.

#define WS2812B_GPIO_CLK_ENABLE() __HAL_RCC_GPIOC_CLK_ENABLE()

Do you have oscilloscope at hand? Do you see some signal on the output or
it is completely dead?

Try enabling the clock and let me know. If you have still issues then send
me the code/project if its not confidential.

Best regards
Martin

alxhoff, 2017-08-22:
Hi Martin,

Big thank you for your very very quick response, I was about to pack it in
for the night but now i've gotten a second wind of hope. It seems I missed
that I needed to modify the GPIO clock define, silly mistake. THe led count
didn't seem to have an effect but I am testing this on a keypad pcb I made
up which is a 3*3 matrix where each row is a seperate RGB row and the third
RGB in each row does not change. Is this a result of the driver?

Thank you agian for this library, great stuff!

Cheers,

Alex

hubmartin, 2017-08-22:
Hi Alex,

I'm glad you have it working. Yes, the last LED is off because the number
of LEDs has to be even. It could be fixed, but I was little lazy. Its
because the transmit cycle always ends in the DMA complete transfer and the
half transfer IRQ only computes buffers and do not checks for the LED
count. So change it to 4, keep the buffer size at 3 and it should work ok.
The data for the 4th LED will be send, but since there is no next LED the
data will be clocked but it wont affect the visible LED.

Best regards
Martin

alxhoff, 2017-08-22:
Hi Martin,

I'll try and get my head around all the stuff that the library is doing
over the next weeks when i get time and see if i can patch that up.

Otherwise, seems to work great now, quickly plugged the otehr 2 rows in and
they are all happily flashing.

Cheers,

Alex

alxhoff, 2017-08-24:
Hi Martin,

Sorry to bother you. I'm home sick for the day so thought i'd look into the
WS2812's again as I wasn't convinced they were working correctly as I
couldn't seem to manually set the colours. And it seems i was right. I have
a strip of 10 LEDs now being powered from a single GPIO and i am only not
getting any responses from the LEDs. I have been through the config stuff a
few times now and cannot see what i'm missing :( I have created a fresh
project without RTOS and no peripherals enabled to see if it was a clash
with something else in my project but it also does not work on the new
project.

I have led's hooked up to PF13 and am running this on a Nucleo144F429ZI at
168MHz. I have also tried a barebones project on a DiscoveryF4 board with
the same results. I feel like with the three seperate LED arrays I have
tried and two boards, I am missing something, hopefully not too obvious :(

I know it is a lot to ask but is there any chance you could have a quick
look at my project and see if you can notice the problem?

Thank you,

Alex

https://drive.google.com/file/d/0B-MzI6MHIvAzR2lMYzlRUzNtcnJ1aHBhQXRLN3VxYm1UQ2Nv/view?usp=sharing

^^project^^

hubmartin, 2017-08-24:
I'll try to take a look at the code.
Which IDE/toolchain do you use?
Which debugger do you use?
Do you have oscilloscope or a logic analyzer?

alxhoff, 2017-08-24:
Hi martin,

Thanks you so much. Sadly I do not. If i was still doing hardware stuff at
university I could maybe use a lab but I no longer have access to any. I
should probably buy myself a logic analyzer though, something I should own.
I am using STM's system workbench and their inbuilt toolchain and debugger.

Very much appreciated,

Alex

alxhoff, 2017-08-24:
Sorry i should of expanded on the workbenches debugger and toolchain.
Debugger is GDB through eclipse  and the toolchain I believe is are eabi
with gcc.

Thanks

hubmartin, 2017-08-25:
What OS you are running Win/macOS/Linux?

Take a look at the MCU pin which sends data. Connect the voltmeter there (or plain LED) to see if the output is logical low, or if ther is some small averaged voltage which means the library is somehow running. Measure and write the measured voltage here.

Make sure and confirm to me that you have enabled code optimizations in your IDE/toolchain. Its explained in the first few lines in the README.md on github.

What frequency is your MCU running? You write 168MHz, please make sure that the function HAL_RCC_GetSysClockFreq() outputs your desired 168 000 000 Hz. Your code seems ok but we need to make sure because if it is running at 8/16MHz then the library don't have to work properly.

By putting a breakpoint in the running program make sure that:
- It never jumps in the DMA_TransferError (put the breakpoint before running the debugger)
- That the library periodicaly calls DMA_TransferHalfHandler
- That the library periodicaly calls DMA_TransferCompleteHandler
- That the library periodicaly calls HAL_TIM_PeriodElapsedCallback

You've mentioned that you own Discovery F4 board. The original GitHub code is a Atollic truestudio project for Discovery F4. Download Atollic (its Windows/Linux now) and import my project from the GitHub and check whether the digital LED strip is running.

alxhoff, 2017-08-26:
Hi Martin,

Thanks for your response. I am truly confused now because I have tried
everything possible. I am on a Linux system, debian based.

Firstly I imported your project into TrueSTUDIO and ran that with two
seperate LED arrays, changing the GPIO clock port, GPIO port, GPIO pin,
number of LEDS,number of LED buffers and the length of the buffers in
visEffect.c.

I was previously using -g optimization but have now also tried with -O1.

There was around 1.3v on the output pin when visHandle was being executed,
I also tried it with three seperate LED arrays, one 3 LED array that is one
row of a test keypad i've made, a 10 LED array that i made from linking the
arrays from another testpad PCB ( I have attached a photo) as well as a 3
LED array of the ws2812 (not b's, but they worked with the avr library due
to the very similar timing). Although I have used this hardware with an AVR
library so I know it works, plus the hardware is very simple :P

The system clock is 168MHz as desired and the code never enters
TransferError whilst periodically entering all the other functions, also as
expected.

The LEDs go from a sort of dull white to a clear white when the code starts
but there is no colour at all. I am totally stumped as I pulled the code
apart and I can't see why it is not working (very elegantly implemented by
the way). I also tried all of this in the STM workbench for both boards
again (Nucleo and discovery).

I am as before at a total loss as to what is missing.

Thank you greatly for your help, I can imagine this is quite annoying to
deal with.

Alex

hubmartin, 2017-08-26:
Hi Alex,
it all seems like the code is working properly. Code jumping to the DMA IRQs and 1.3v on the output seems like the MCU sends data. I can't see the picture in the attachement you've mentioned.

I was worried if you don't have damaged output pins of your MCU but 1.3 sounds like it is working.

You've said that with AVR the LED strip is working. AVR chip is 5V, STM32 is 3V, did you use the level shifter for the WS2812 data signal? WS2812 expects 5V data signal. It may work with 3V signal on your table, but in final product it could be an issue as I learned once. You can get without the level shifter with a small hack but then you have to power the LED strip with lower voltage (4-4.5V).. it could be run from 3 V maybe to test it. But I can imagine that there is few types and manfacturers of WS2812(B) chips with on the edge specifications.

I have fresh install of Windows 7 so I've cloned my project, compiled With Atollic 7.0.0 and run on STMF4 Discovery. Connected the LED strip to GND, 5V VDD and PC0 and it works. [Default working firmware with connection](https://goo.gl/photos/k3E2zqtuC8A6depC8)

I'm a little confused whether you compiled and run my Github code without a single edit, or did you everytime changed the configuration for your pins? There is no need to change the code even if you have lower count of LEDs. Just clone, compile, run of F4 Discovery (the github Atollic project has -Og optimizations enabled).

IF you don't have logic analyzer, you have to build one with your arduino. Try to google a little. What I  would search is implementation of SUMP protocol with opensource Sigrok PC software. Or there's many other examples like https://github.com/gillham/logic_analyzer but I haven't used any of them.

Martin

alxhoff, 2017-08-26:
Hi Martin,

I tried a completly fresh copy of you code on both the nucleo and discovery
since they share the same MCU and both gave weird results (see video
https://goo.gl/photos/RVZkmqtzRX2mEU4E9), this makes me think it's a
hardware issue. Quit possibly what you have mentioned with the stm only
driving 3.3v. The data sheet says -0.5～VDD+0.5 for the input voltage so i
was assuming the 3.3 would do the trick but I cannot think of what it could
be otherwise. I think i'll get a logic level converter, logic analyzer
(cannot put this off any longer) as well as a few more differnt LED strips
to test with and see what happens. So hopefully in a month you'll recieve a
pretty picture of my LEDs working. But for thank you so much for your help.
The open source/nerd community needs more people like you.

Big thank you!

Alex

hubmartin, 2017-08-26:
Lol, the white blinking is the actual effect. So congratulations :Đ It's for christmass.. white snowflakes :)
On the PC0 & PC2 there's a rainbow affect and on the PC1 & PC3 there's the effect you have on the video.
In the previous post you mentioned that you see some dull white color when the code starts but I was thinking it is white before the code starts. So I was not giving that any attention.

hubmartin, 2017-08-26:
You definetely need the level shifter. It don't worth the issues you can then have later.

alxhoff, 2017-08-26:
HI Martin,

Glad that it was the efffect. It only happens randomly. Maybe one out of
every few tries does it actually work and the rainbow didn't. Makes me more
certain that it is the logic level. Have just ordered some level shifters
so i'll find out in about a month. You dind't use a series current limiting
resistor on Din did you? Online there are a few people using them but I
feel like the internal resistance of these things should do it and I don't
want to drop my input voltage any more than it already is :(.

Hopefully it's just he logic levels!

Thanks,

Alex

alxhoff, 2017-08-26:
If i power the LED's from an external source and then only drive Din from
the board the dots work. Must be too much for the board. THe rainbow
doesn't work so i think it's the voltage :(

hubmartin, 2017-08-26:
As I said, you can power the WS2812 from 3V power rail directly from Discovery.. This way you fit in the 0,5-VDD+0,5 DIN voltage limit. It works right now on my table. So you can find if the reason that sometimes it don't work is the DIN voltage.

You say that the rainbow effect don't work. That's weird, I connect DIN to PC0 on my discovery with Github code and there's a rainbow. If it displays the snowflake effect on PC1 then there's not a reason why it wouldnot display rainbow effect on PC0.

alxhoff, 2017-08-26:
Ahhh that makes a lot of sense now :P brain fart. Eurika! the 3volts drives
it well. I will now have to see if my projects work with the code.

Thank you so so much!
https://goo.gl/photos/h9g5XXdDxwX7zPZj7

hubmartin, 2017-08-26:
Can you share some more info about your project? I guess it is not opensource, but I'm interested in where my code would be :) It seems like color keypad to me. Can I use your video or picture on my Twitter to show others how people use my library? Do you use twitter? OR you can post it and mention me @hubmartin

alxhoff, 2017-08-26:
I sure can share it with you. It's not open source in that i've only
started actually coding the projects. it's a mechanical keyboard revolving
around a cortex m4 so that more powerful things can be added into it at a
later date. But also i have a design where the leds are recessed into the
pcb so that each switch can have an led. and since these leds are each
individually controllable that leads to some powerfull led stuff. But
mainly becuase the smaller 8 bit controllers struggle to run a keyboard and
led's etc all at once with overhead left over to do other stuff. I do a lot
of linux work and it would be nice to have scripts stored in my keyboard so
that if i plug my keyboard into a server it can "download" the script by
esentially typing it into vim. Things like this.

You can use my picture no worries but wait a seccond and ill get you a much
better one.

At the moment it's just a heap of dirty code that i wrote quickly to get
the basic functionalty done. But i need to add all the other more advanced
yet still basic features like macros and multiple layers. I dont have
twitter but i'd be more than happy to share the project with you once i get
it to a level where I am happy enough to share it :) I use my own private
git servers but ill put it onto github at some point.

Enjoy the video and picture.

Ill keep you updates about the project. About to send a full size keyboard
test pcb to get made. Will be cool to see your library driving ~70 LEDs.

Alex

https://goo.gl/photos/aWMj7szqJEa4AkQR7

https://goo.gl/photos/Sh5jkFJWKAd5uShY6

alxhoff, 2017-08-26:
This is the design that recesses the LED into the board.

This is the test board that will plug into a Nucelo for development.

https://goo.gl/photos/zUHQAkgYagNtj9Ph9

alxhoff, 2017-08-26:
Just tried it with my RTOS project with the keyboard controller code, work
a charm! only If i change the pinout to use 0,1,2 and 13 on port F the pin
13 doesn't work. The entire port is otherwise free.

hubmartin, 2017-08-26:
Great, if you have issues with stability with RTOS make sure that DMA IRQs has the highest priority, which is the lowest number value on ARMs.
Thanks for the images, both keyboards look awesome. I like that buttons.
I'll close this thread, please ping me your mail to to hub.martin _at gmail _dot_ _com_ so you can send me your progress later.

closed 2017-08-26 by hubmartin