hubmartin/WS2812B_STM32F4 #4: Changing to TIM3 and PIN PC7

cmdc0de, 2018-07-30:
Per our twitter conversation:
///
Hi. It's lot of things that needs to be changed when you choose another timer. My guess is that you need the TIM2 to some specific task/PWM.
Please start the Github issue and post the source code. Thanks, we will solve the problem there.
///
except I actually need to use TIM 3

attached are the changes I made, if I just change line 94 to TIM1, uncomment lines 103 and 104 things work.
change 94 to TIM3 comment lines 103 and 104 and uncomment 105 and 106 and they don't.  Not sure I understand why.  All other changes in the files are to use PC7 and remove the hardcoded use of TIM1-> type stuff.  

Thanks for all the help in advance!!!!!!
[STM32F4_WS2812B.zip](https://github.com/hubmartin/STM32F4_WS2812B/files/2239854/STM32F4_WS2812B.zip)

hubmartin, 2018-07-30:
Hi, you need to change a lot of things.

First read this issue which explains in the picture/table that you have to change DMA channels/streams when you use different timer. https://github.com/hubmartin/ws2812b_stm32F3/issues/1
You have to use DMA2 because DMA1 don't have access to the GPIO bus.
Also you have to change NVIC commands and Interrupt service routine names to the right DMA Channel.
Do you have a debugger? The best check is put breakpoints to tha DMA ISRs (DMA Hlaf transfer and DMA Full transfer) to make sure that the timer and the DMA works correctly.

Hmm, bad news, according to DMA2 request mapping table there is no any other timer in this table. And DMA1 cannot be used for reasons above. So I'm afraid it could not be used with any other timer.
[image]

Then the switch to PC7 is easy:
Set only PIN7 in this array
https://github.com/hubmartin/STM32F4_WS2812B/blob/master/Src/ws2812b/ws2812b.h#L24

And here in the array you set the "channel" to value 7 so its driving pin 7
https://github.com/hubmartin/STM32F4_WS2812B/blob/master/Src/visEffect.c#L136

Something like:
```
ws2812b.item[0].channel = 7;
ws2812b.item[0].frameBufferPointer = frameBuffer;
ws2812b.item[0].frameBufferSize = sizeof(frameBuffer);
```

Try to apply these edits. If you could not get it running, append your new code and I'll take a look at your edits.
Martin

cmdc0de, 2018-07-30:
Thank you!!!! the mcu I'm using is a STM32F411RE.  If I'm reading the DMA mapping correctly I could use TIM1 as long as I can use GPIOC and Pin 7 (board is already completed).  However I need to TIM1 to run off the internal clock and not the external crystal due to pin conflicts. I have attached my ioc file so you can see.
[darknet-7.zip](https://github.com/hubmartin/STM32F4_WS2812B/files/2240236/darknet-7.zip)

hubmartin, 2018-07-30:
Then you would need to use some other library on F4. There is no other timer in the DMA2 table.
I see that pin PC7 has tome other timers. Maybe you can try to port my other library for STML0.

THe F4 and F3 libs in my Github is using DMA > GPIO transfers and main advantage is that you can simple drive all 16 bits in single port with really low memory footprint. However the STM32L0 don't have DMA > GPIO capability so I changed the signal generation to the timer compare PWM.

So if you are using only one output for single ws2812 digital LED strip, the STM32L0 implementation could help you. But you need to port it yourself, configure PWM on the timer, set correct DMA1 channels/streams.
https://github.com/hubmartin/WS2812B_STM32L083CZ

Please check also the DMA > TIMer section of my page 
http://www.martinhubacek.cz/arm/improved-stm32-ws2812b-library

Martin