hubmartin/WS2812B_STM32F4 #7: DMA handlers not called on STM32F446RE

nklz, 2020-04-03:
I have a fairly simple setup with 6 parallel outputs. However I have ran into problems with LEDs either not showing up or flickering or similar things. First thing I tried was hooking up an logic analyzer to one of the data pins, which shows data sent with correct timing, but seemingly the same 8 bits repeatedly and no reset signal.
Further investigation showed that I `ws2812b.transferComplete` is never reset after beginning transmission. Since that is done in the DMA callbacks, I decided to set breakpoints in the IRQ and the HAL handlers and noticed that while the IRQ is called, the HAL handlers are never called.
I didn't really go into detail about how this library really works and therefore do not know whether its relevant or not, but the Timer callbacks and the Timer IRQ are never called aswell.
I would appreciate help getting this to work.

nklz, 2020-04-03:
I have had a look at the HAL DMA code and it appers that the condition `__HAL_DMA_GET_IT_SOURCE(hdma, DMA_IT_HT) != RESET` is not met and therefore no callbacks are called.

nklz, 2020-04-03:
I have found the issue. HAL_DMA_DeInit resets all values in the DMA_HandleTypeDef passed to it. Therefore no values and callbacks are stored (and the interrupt flags are not set in HAL_DMA_Init). Commenting these lines fixed it. I'd suggest doing so in the code provided here aswell if there is no real purpose for these.

jcrademacher, 2020-07-21:
@nklz I noticed the transferComplete flag wasn't being reset as well. Thanks for noticing this, I've been struggling with this for a while now. I'm commenting here so hopefully martin sees this and gets rid of those lines for others who want to use the library.

hubmartin, 2020-07-21:
Hi, thanks for info. Can you test this PR #8 ?
More details in the PR.
Can someone with hardware and correct compiler/IDE test it? 
Thanks

hubmartin, 2020-07-21:
So it motivated me enough to build some hardware with Discovery kit and LED strip. I've updated the HAL from Github and the #8 fixed it.

jcrademacher, 2020-07-23:
I've tested PR #8 and it works correctly.

hubmartin, 2020-07-23:
Thanks for confirmation.

closed 2020-07-23 by hubmartin