hubmartin/WS2812B_STM32F4 #9: Support for STM32F7

madnat, 2020-07-26:
Hi there - really enjoyed your material, helps me understand the bit banding features of this processor family.

Can you tell me which libaray would best support the STM32F7 processor?  Thanks you

hubmartin, 2020-07-26:
Hi, someone tried my F4 version of F7 but did not shared more info or code so I couldn't help with that.
What I remember is that F7 because of caches does not have bit-badning and you have to use less effective bit-juggling variants like `SETPIX_3`.
Some info about porting to different family and how to check timers/DMA is explained here https://github.com/hubmartin/WS2812B_STM32F3/issues/1

hubmartin, 2020-07-26:
I've received the code from the guy who did porting to F7. You can start from here, he says it should work.
[stm32f7.tar.gz](https://github.com/hubmartin/WS2812B_STM32F4/files/4978294/stm32f7.tar.gz)

madnat, 2020-07-27:
Thanks - thats really very helpful - and appreciate you getting the code.  It compiles with no issue - although your sample visEffects does not run (just white constant LED's from Pin PD1).  

As you suggest - configuring TIM and DMA is required for the F7 chip - looks beyond my limited knowledge on how to do that - reading the posts from F3 content you mention - its unclear how that would help me setup for a F7.  Thanks again for sharing the code - I will try to hack my way through setup of TIM and DMA.  Any pointers/articles or resources you feel will help would be very helpful.

hubmartin, 2020-07-28:
You're welcome.
That F3 > F4 tables I sent the link showed the relation betweet the timer you use and the DMA channels for the UP, compare 1 and compare 2 DMA streams/channels. You have to find this table in F7 reference manual, might be different, might be same.. not sure.

madnat, 2020-07-29:
I had a good search for the documentation - but couldn't find the corosponding tables.  However, using STMCubeIDE, I was able to determine what I think is a correct configuration.  To me it looks like DMA1, Stream 1, 5 & 6 support PWM Generation on TIM1 UP/CH3, CH1 & CH2/CH4 respectively.  Do you Agree?
[image]
[image]

hubmartin, 2020-07-29:
You are talking about TIM1, but in DMA1 there is no TIM1 and you are refering to TIM2.

I've opened the code I've sent you and it seems configured fine.
Check Table 28 in reference manual to your chip
https://www.st.com/resource/en/reference_manual/dm00224583-stm32f76xxx-and-stm32f77xxx-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf

The user of F7 who sent me the code used TIM1, see function `static void TIM1_init(void)`
He used DMA2 which has streams and channels for TIM1_UP, TIM1_CH1 & TIM1_CH2

[image]

Then in `static void DMA2_init(void)` function all this seems all set-up correctly `DMA_CHANNEL_6` with `DMA2_Stream5` for TIM1_UP.
`DMA2_Stream1` for TIM1_CH1 and `DMA2_Stream2` for TIM1_CH2.

So check if your code is jumping to the TIM1 IRQ handler `TIM1_UP_TIM10_IRQHandler`.
Then check if the code jumps to apropriate DMA IRQ handler `DMA2_Stream2_IRQHandler`.
If not, you have to fix it first.

madnat, 2020-07-29:
Sorry - I was confused by the different permutations!  You are absolutley correct I meant TIM2 on DMA1  (I edited my post to correct the typo).

I am now looking at TIM1 on DMA2 (see how easy it is to get turned around!).  Thank you for the pointer to the documentation - I spent a few hours looking for it (and getting distracted learning new things on the way :D).

I have gone back to the original code from the other F7 user you kindly provided.  I will focus on getting a basic version running and move from there.  Really appreciate you helping me.