hubmartin/WS2812B_STM32F4 #10: Help with code on Nucleo-F410RB

splinepd, 2020-12-28:
I'm trying to use your code with a Nucleo-F410RB (100MHz).  I just have one strip of 3 LEDs so far, to verify that things work.  I have the LEDs directly connected to PB3 and most everything ported fairly easily.  The LEDs are all white after init and while running.  I have an oscilloscope on the output pin and the frequency and duty cycle looks good.  The only place WS2812_RESET_PERIOD shows up is in HAL_TIM_PeriodElapsedCallback(), but it's commented.  Any ideas?  

BTW, I tried reaching out on your website, github, and Twitter, but couldn't find an email address and Twitter DMs are off.

hubmartin, 2020-12-29:
The code has some commented parts because it was developed over time for F1, F3, F0, L0.. see my other repos.
The macro is commented because the 50 us reset pulse is handled diferently in this F4 port.
When the last LED data is sent the timer is stopped, reconfigured to 50us and run again. Also this 50us period is calculated dynamically.
https://github.com/hubmartin/WS2812B_STM32F4/blob/master/Src/ws2812b/ws2812b.c#L359

sahilrastogi94, 2021-04-11:
@splinepd Hey, I am also trying to run this on my Nucleo-F410RB but couldn't seem to get it to work. I have previously ran this on F407 discovery board and Nucleo-F302R8 board without any problems. I am using CubeMX to configure my board. Not sure why it is not working. Do you think you can share your project with me? Any help would be much appreciated.

splinepd, 2021-04-13:
Sorry, I haven't had a chance to get back into this.