hubmartin/WS2812B_STM32F4 #13: STM32F072 Port

Parcival97, 2021-04-16:
Hello, first of all thank you very much for the post and for the WS2812B_STM32F4 library. I am currently trying to make a port for the STM32F072 but I am struggling trying to understand the timer period, reset pulse and cc1 and cc2 register lines, in the ws2812b.c file. More precisely, they are inside the TIM1_init() function. The lines are the folllowing ones:

tim_period =  SystemCoreClock / 800000; // 0,125us period (10 times lower the 1,25us period to have fixed math below).
timer_reset_pulse_period = (SystemCoreClock / (320 * 60)); // 60us just to be sure. The resetpulse must be >50us.

//The next two values are used to specify the pulse value to be loaded into the Capture Compare Register.
uint32_t cc1 = (10 * tim_period) / 36;
uint32_t cc2 = (10 * tim_period) / 15; 

I don't understand how the timer period is 0,125us. If I make the operation tim_period = 16MHz/800000 = 20.
I also don't understand the operation made in reset pulse period. I cannot see why the result is
timer_reset_pulse_period = 16MHz / (320*60) = 833,333333.

The same happpens with cc1 and cc2 in which I don't undesrtand why tim_period is multiplied by 10 and then divided by 36 or 15.

I would be very grateful if you could help me with this problem and explain me this in detail. I would also like to know if you have any port made STM32F0 series. It could also help me to better understand some configurations. I am new to this world and I am still learning so any help would be really appreciated!!

Thank you very much and I hope you can answer me with this issue.

Best regards,

hubmartin, 2021-04-17:
Hi, there is no port for F0, it seems I have it wrong in few readme files.
I try to answer from top of my head, its really old project, I don't have any hardware nor time to test it.
My first guess is that 16 MHz could be too low. Many years ago I ported this to F100 which runs at 24 MHz and I had to overclock it to 48 MHz to get rid of jitter in the generated data.
I would suggest to port and compile the code and look at the outputs with scope/logic analyzer to see what really comes out the pin. This is sometimes needed to see if the DMA/CPU is fast enough in the end. This code is heavily dependend on DMA/hardware and there is no other way to confirm funcionality or debug it without scope/logic analyzer.

The timer increments every `tim_period` which is 0.125 us on F4.
From this period you get all other values.
`cc1 = (10 * 0.125) / 36 = 0,0347` which corresponds to 350ns (in the image below)
`cc2 = (10 * 0.125) / 15 = 0.0833` which corresponds to 700ns (it is longer to make sure the LED get 1 right in case some DMA jitter)

[image]

Check the image on my page
http://www.martinhubacek.cz/arm/improved-stm32-ws2812b-library

Parcival97, 2021-04-19:
Thank you very much for the soon answer. I really appreciate it. I will try to do as you say. Thanks again!