hubmartin/WS2812B_STM32F4 #16: STM32F407 Running 16 strips, 100leds/strip. Flicker

PWawrow, 2023-03-11:
Hello. I get random flickering leds 5-6 for per strip, other seem to work fine. Maybe something with timing? Hardware: STM32F407VET6 SYSCLK-168MHz APB2 TIMER CLOCK-168MHz. I'm using rainbow example from library.

hubmartin, 2023-03-12:
Try to use a different SETPIX option. Lower numbers are less effective. Test it and see whether the issue stays or gets worse.
Try to lower a number of parallel strings. If it helps, then the DMA interrupts for some reason is not fast enough to fill the buffers on the fly.
Try to look with the oscilloscope/logic analyzer to see what happens on that fifth or sixth LED and whether the issue happens at the same time on multiple outputs.
Check whether the DMA IRQs have the highest priority if you have added other code using IRQ (probably not if you use just the example code)

hubmartin, 2023-03-12:
Which compiler optimization level do you use? -O.. ?